jaws/jaws #10: Create daily, monthly, yearly, climatological plots of Summit data

czender, 2017-12-27:
Create four nice plots of the summit data. Thumbnail them and add them to the homepage. The daily plot just shows one day of raw data. The monthly plot shows the full diurnal cycle of all thirty days, along with the daily mean, daily max, and daily min temperatures. The yearly plot shows monthly means with standard deviation of the daily-mean data within each month plotted as an envelope. The climatological seasonal cycle plot shows same, except standard deviations created for each month.

ajcse1, 2018-01-03:
What does 'summit' mean here? Does it mean the raw data that we have or is it any other dataset?

czender, 2018-01-03:
It means the GCNet station at Summit.

ajcse1, 2018-01-14:
#15 Done

closed 2018-01-14 by ajcse1