jaws/jaws #11: Add output options for different netCDF formats

czender, 2017-12-27:
Add -3, -4, -5, -6, and -7 (short) and corresponding long_name output options with same names as NCO's --fl_fmt options to allow the user to choose netCDF output format.

czender, 2018-01-05:
Also, I mean support the same exact argument formats as NCO, to wit, these and similar should work as documented in the NCO manual:
```
python jaws.py -3 in.txt
python jaws.py --3 in.txt
python jaws.py --fl_fmt=64bit_offset in.txt
python jaws.py --fl_fmt=cdf5 in.txt
python jaws.py --fl_fmt=netcdf4 in.txt

```

czender, 2018-01-09:
Just tried again and the format handling does not work for netCDF4, much less netCDF3 or other non-defaults:

```
zender@aerosol:~$ python ~/jaws/source/jaws.py -4 ~/jaws/sample_data/promice_20160501.txt -o ~/promice_20160501.nc
usage: jaws.py [-h] [-o OUTPUT] [-3 [FORMAT3]] [-4 [FORMAT4]] [-5 [FORMAT5]]
               [-6 [FORMAT6]] [-7 [FORMAT7]]
               input
jaws.py: error: argument -4/--format4/--4/--netcdf4/--fl_fmt=netcdf4: invalid int value: '/Users/zender/jaws/sample_data/promice_20160501.txt'
zender@aerosol:~$ python ~/jaws/source/jaws.py --4 ~/jaws/sample_data/promice_20160501.txt -o ~/promice_20160501.nc
usage: jaws.py [-h] [-o OUTPUT] [-3 [FORMAT3]] [-4 [FORMAT4]] [-5 [FORMAT5]]
               [-6 [FORMAT6]] [-7 [FORMAT7]]
               input
jaws.py: error: argument -4/--format4/--4/--netcdf4/--fl_fmt=netcdf4: invalid int value: '/Users/zender/jaws/sample_data/promice_20160501.txt'
zender@aerosol:~$ python ~/jaws/source/jaws.py --fl_fmt=netcdf4 ~/jaws/sample_data/promice_20160501.txt -o ~/promice_20160501.nc
usage: jaws.py [-h] [-o OUTPUT] [-3 [FORMAT3]] [-4 [FORMAT4]] [-5 [FORMAT5]]
               [-6 [FORMAT6]] [-7 [FORMAT7]]
               input
jaws.py: error: argument -4/--format4/--4/--netcdf4/--fl_fmt=netcdf4: invalid int value: '/Users/zender/jaws/sample_data/promice_20160501.txt'
zender@aerosol:~$ python ~/jaws/source/jaws.py --format4 ~/jaws/sample_data/promice_20160501.txt -o ~/promice_20160501.nc
usage: jaws.py [-h] [-o OUTPUT] [-3 [FORMAT3]] [-4 [FORMAT4]] [-5 [FORMAT5]]
               [-6 [FORMAT6]] [-7 [FORMAT7]]
               input
jaws.py: error: argument -4/--format4/--4/--netcdf4/--fl_fmt=netcdf4: invalid int value: '/Users/zender/jaws/sample_data/promice_20160501.txt'
zender@aerosol:~$ python ~/jaws/source/jaws.py -3 ~/jaws/sample_data/promice_20160501.txt -o ~/promice_20160501.nc
usage: jaws.py [-h] [-o OUTPUT] [-3 [FORMAT3]] [-4 [FORMAT4]] [-5 [FORMAT5]]
               [-6 [FORMAT6]] [-7 [FORMAT7]]
               input
jaws.py: error: argument -3/--format3/--3/--fl_fmt=classic: invalid int value: '/Users/zender/jaws/sample_data/promice_20160501.txt'
zender@aerosol:~$ python ~/jaws/source/jaws.py --3 ~/jaws/sample_data/promice_20160501.txt -o ~/promice_20160501.nc
usage: jaws.py [-h] [-o OUTPUT] [-3 [FORMAT3]] [-4 [FORMAT4]] [-5 [FORMAT5]]
               [-6 [FORMAT6]] [-7 [FORMAT7]]
               input
jaws.py: error: argument -3/--format3/--3/--fl_fmt=classic: invalid int value: '/Users/zender/jaws/sample_data/promice_20160501.txt'
zender@aerosol:~$ python ~/jaws/source/jaws.py --fl_fmt=classic ~/jaws/sample_data/promice_20160501.txt -o ~/promice_20160501.nc
usage: jaws.py [-h] [-o OUTPUT] [-3 [FORMAT3]] [-4 [FORMAT4]] [-5 [FORMAT5]]
               [-6 [FORMAT6]] [-7 [FORMAT7]]
               input
jaws.py: error: argument -3/--format3/--3/--fl_fmt=classic: invalid int value: '/Users/zender/jaws/sample_data/promice_20160501.txt'
zender@aerosol:~$ python ~/jaws/source/jaws.py --format3 ~/jaws/sample_data/promice_20160501.txt -o ~/promice_20160501.nc
usage: jaws.py [-h] [-o OUTPUT] [-3 [FORMAT3]] [-4 [FORMAT4]] [-5 [FORMAT5]]
               [-6 [FORMAT6]] [-7 [FORMAT7]]
               input
jaws.py: error: argument -3/--format3/--3/--fl_fmt=classic: invalid int value: '/Users/zender/jaws/sample_data/promice_20160501.txt'

```

ajcse1, 2018-01-09:
If we are passing the format argument after the input argument, then it is working fine. e.g.:

`python ~/jaws/source jaws.py ~/jaws/sample_data/promice_20160501.txt -4 ~/promice.nc`

or 

`python ~/jaws/source jaws.py ~/jaws/sample_data/promice_20160501.txt ~/promice.nc -4`

I am working on to fix why it is throwing error if we pass it before the required input argument. Thanks for pointing this out.

czender, 2018-01-09:
OK. The reason to "fix" it is because we want to support canonical UNIX ordering which is:
command [options] [input] [output]

czender, 2018-01-09:
The problem was that I used [-o output] (which is an option) _after_ [input]. This does work:
`python ~/jaws/source/jaws.py --fl_fmt=classic  -o ~/promice_20160501.nc ~/jaws/sample_data/promice_20160501.txt` but this does not:
```
zender@aerosol:~$ python ~/jaws/source/jaws.py --fl_fmt=classic ~/jaws/sample_data/promice_20160501.txt
usage: jaws.py [-h] [-o OUTPUT] [-3 [FORMAT3]] [-4 [FORMAT4]] [-5 [FORMAT5]]
               [-6 [FORMAT6]] [-7 [FORMAT7]]
               input
jaws.py: error: argument -3/--format3/--3/--fl_fmt=classic: invalid int value: '/Users/zender/jaws/sample_data/promice_20160501.txt'
```
When an output filename is not given it should default to the input name, in the working directory (not the input file directory), with a .nc suffix.

ajcse1, 2018-01-09:
The arguments can now be passed in any order. When the output filename is not given, it is defaulting to the input name in current working directory with a .nc suffix.

ajcse1, 2018-01-18:
All the datasets can now be converted to netCDF classic formats with the options at command line.

closed 2018-01-18 by ajcse1