jaws/jaws #4: Add sample one-day dataset of IMAU raw data

czender, 2017-12-27:


ajcse1, 2018-01-17:
I have contacted Carleen to get the data. She forwarded my request to Paul Smeets (Data Manager). Paul asked me on 01/12/18 about the data interests i.e. area/locations, years, parameters and  time resolution. I replied on the same day requesting him to provide data of all the parameters for all the locations since the beginning of AWS.

ajcse1, 2018-01-31:
Paul replied and said that he will send all raw Greenland AWS data from type II AWS (2004 - 2016) and from Antarctica (1997 - 2015 multiple stations) this week. Station II means that for these stations the same set of sensors was used during the entire period.

ajcse1, 2018-02-16:
Added each for Greenland and Antarctica stations

closed 2018-02-21 by ajcse1