jaws/jaws #6: Delete the multi-megabyte sample data

czender, 2017-12-27:
We do not have permission to re-distribute large amounts of raw data. Moreover, we want to keep the JAWS repository lightweight. We can include one day of sample data from each network for starters. Delete and permanently remove from GitHub repository the existing multi-megabyte sample datasets assuming they exceed one-day in length. Those data should only be on our private machines, not in the repository.

czender, 2018-01-02:
What command did you use to delete the monthly data? The repository is still 3 MB in size and I'm unsure why. There is not that much source code. Please completely expunge the monthly data so they do not take up space in the repository, and are unrecoverable. 

```
zender@aerosol:~$ git clone https://github.com/jaws/jaws.git
Cloning into 'jaws'...
remote: Counting objects: 225, done.
remote: Compressing objects: 100% (96/96), done.
remote: Total 225 (delta 84), reused 94 (delta 41), pack-reused 88
Receiving objects: 100% (225/225), 2.95 MiB | 70.00 KiB/s, done.
Resolving deltas: 100% (102/102), done.

```

ajcse1, 2018-01-03:
I used git rm to delete the data. There is a file 'AWS14_SWdata_Wenshan.txt' in the original_scripts/dat_sample directory provided by Wenshan having size of 2.4 MB. Should I remove it?

czender, 2018-01-03:
Re-read my request. This is what I mean by "completely expunge":
https://help.github.com/articles/removing-sensitive-data-from-a-repository/

We have no right to put others' data in a public repository without first asking. 
Yes, please expunge the 2.4 MB file from Wenshan. 
Add as many one day test files as are necessary to completely test the software.
Tests on longer datasets (monthly, yearly) should be performed on local data that is not in the repository, and should be in a standard location (e.g., $DATA/jaws) on, say, katabatic and jaws.
Please think twice before adding any large files. 
If you typed it (i.e., source code, documentation), you can add it, otherwise ask me.

ajcse1, 2018-01-05:
Thanks for the link. Completely expunged the files. The repository is less than 1 MB now.

closed 2018-01-05 by ajcse1