jaws/jaws #9: Add GPS-derived ice velocity

czender, 2017-12-27:
For stations that archive GPS position, diagnose the ice velocity.

ajcse1, 2018-01-28:
Calculated for PROMICE using Haversine formula. The new variable is named as ice_velocity_GPS

What standard name should we give to this variable? I looked in the CF table and found the following variables:
land_ice_surface_x_velocity
land_ice_x_velocity
land_ice_basal_x_velocity

czender, 2018-01-29:
Do you have total velocity or x and y components? It is fine to store all three (x,y,total).
And use land_ice_surface_x_velocity and land_ice_surface_y_velocity.

czender, 2018-01-29:
Please repeat twice, and add x and y velocities. Positive x is towards east, positive y is towards north. Consider adding "positive" CF attribute to the velocity variables.

ajcse1, 2018-02-12:
Added both x and y components.

closed 2018-02-12 by ajcse1