jaws/jaws #14: Create local data repository for all AWS data

czender, 2018-01-03:
As discussed in #6, create a repository in katabatic.ess.uci.edu:/Users/sainiak/data/jaws and in the corresponding directory on jaws.ess.uci.edu. Here $DATA=/Users/sainiak/data and we will refer to the local repository as $DATA/jaws on any machine where it exists. The contents of $DATA/jaws will be subdirectories promice, gcnet, aaws, etc. Each of these subdirectories will have its own subdirectories named raw and nc. Place the raw data in raw, and we will use nc for the processed data. The processed data will "periodically" be deleted and recreated by jaws.py so that all datasets in nc will reflect the latest JAWS features. Periodically means roughly weekly, based on that week's new features, or whenever a new point version is released.

czender, 2018-01-16:
`for fl in `ls ~sainiak/data/jaws/promice/raw/*_hour.txt`; do echo $fl; python ~/jaws/source/jaws.py $fl -o ${fl/txt/nc}; done`

czender, 2018-01-16:
I installed Miniconda3 (i.e., Python 3.6 not 2.7) in my directory on katabatic. Now I think the following command should work, but it never does:
```
zender@katabatic:~/data/jaws$ for fl in `ls ~sainiak/data/jaws/promice/raw/*_hour.txt`; do echo $fl; python ~/jaws/source/jaws.py $fl; done
/Users/sainiak/data/jaws/promice/raw/EGP_hour.txt
  File "/Users/zender/jaws/source/jaws.py", line 26
    print "The file " + str(args.input) + " is converted into netCDF format."
                    ^
SyntaxError: invalid syntax
/Users/sainiak/data/jaws/promice/raw/KAN_B_hour.txt
  File "/Users/zender/jaws/source/jaws.py", line 26
    print "The file " + str(args.input) + " is converted into netCDF format."
...
```

ajcse1, 2018-01-19:
Can you please check if it works now? I have made the print statement compatible with Python 3.6

czender, 2018-01-19:
```
zender@katabatic:~$ for fl in `ls ~sainiak/data/jaws/promice/raw/*_hour.txt`; do echo $fl; python ~/jaws/source/jaws.py $fl -o ${fl/txt/nc}; done
/Users/sainiak/data/jaws/promice/raw/EGP_hour.txt
Traceback (most recent call last):
  File "/Users/zender/jaws/source/jaws.py", line 2, in <module>
    import gcnet2nc
  File "/Users/zender/jaws/source/gcnet2nc.py", line 958
    month[n] = 01
                ^
SyntaxError: invalid token
/Users/sainiak/data/jaws/promice/raw/KAN_B_hour.txt
Traceback (most recent call last):
  File "/Users/zender/jaws/source/jaws.py", line 2, in <module>
    import gcnet2nc
  File "/Users/zender/jaws/source/gcnet2nc.py", line 958
    month[n] = 01

...
```

ajcse1, 2018-01-22:
Corrected the above error.

All files for PROMICE and AAWS have been converted on 'grele' using Python2.7. GCNET conversion under progress. 

Installed Python 3.6 on katabatic but required permission to install netCDF4. Asked Matt for permission.

ajcse1, 2018-01-24:
Code compatible with Python 3.6 as well

closed 2018-02-05 by ajcse1