jaws/jaws #15: Create four plots of representative timeseries data

czender, 2018-01-03:
We'll use Summit data for these demo plots:
a) monthly diurnal cycle x=local hour of day (0..23) y=monthly mean values for that hour, indicate variance in hourly data
b) monthly values: x=day of month (1..30), y=mean daily values, indicate max/min for each day
c) annual cycle: x=day of year (1..365), y=mean daily value, shading envelop with daily max/min
d) climatological seasonal cycle: x=month (1..12), y=climatological mean of month, indicate standard deviation of all months.

Please place scripts in new /analysis directory, and plots themselves on the homepage as examples of what can be done with JAWS data. Keep in mind that we may transition to matplotlib, so separate the analysis from the plotting in the scripts as much as possible.

ajcse1, 2018-01-14:
Created all four plots and scripts for these are placed in /analysis directory

ajcse1, 2018-01-16:
Remove grids for all plots
First curve should be black.

For diurnal cycle:
1. plot std deviation on the top of mean. 
2. Horizontal axis- Local hour of day
3. Vertical axis- remove in and put Kelvin and square brackets

For plot 2:
1. Title: remove 'monthly change in '. 
2. Jan-2014 can be automated from script.

For plot 3:
1. Title: temperature at summit for 2014

For plot 4:
1. title: Climatological temperature change at summit (start year - end year)
2. Change months number to characters i.e. 'Jan'

ajcse1, 2018-01-22:
Processed all the above mentioned changes except that in diurnal cycle, the current time is in UTC. It will be converted to local time.

ajcse1, 2018-02-13:
Added functionality to choose timezone from command line.

closed 2018-02-14 by ajcse1