jaws/jaws #17: Document required software stack for Miniconda

czender, 2018-01-04:
A user who follows the README instructions with miniconda will receive this error:
```
zender@aerosol:~/jaws/source$ python jaws.py ../sample_data/gcnet_20130101.txt
Traceback (most recent call last):
  File "jaws.py", line 3, in <module>
    import promice2nc
  File "/Users/zender/jaws/source/promice2nc.py", line 3, in <module>
    from astropy.io import ascii
ImportError: No module named astropy.io
```
Please update README with pre-requisites to conda-install before attempting the examples.

ajcse1, 2018-01-05:
Thanks for pointing this out. Updated the instructions.

czender, 2018-01-05:
Close issues (like this one) that you think have been fully addressed. That is the default. If you want me to review an issue to see whether I think it should be closed, ask me.  Use your judegement.

closed 2018-01-05 by czender