jaws/jaws #22: Make default output name is same as input name with .nc suffix

czender, 2018-01-05:
For example, the output of this command
`python jaws.py ../sample_data/gcnet_20130101.txt` should be a file in the working directory named gcnet_20130101.nc.

ajcse1, 2018-01-05:
I did this by taking the file name as input(gcnet_20130101.txt) and splitting them on '.' and then retaining the first part(gcnet_20130101). If the input file is in the same directory as jaws.py then it's working fine, but if we have the data in another directory and we give the input as (../sample_data/gcnet_20130101.txt), then there is a '.' in the starting and input will be split on that. So, it will not get proper file name.

czender, 2018-01-05:
Then change the algorithm you use because it is incapable of handling important cases. I'm sure Python has innumerable filename-mangling modules. It is part of your job as a professional programmer to know or learn these things as needed, preferably but not always without asking me how, and _certainly without redefining the parameters of the request to fit within your existing knowledge_. I suggest looking for answers and asking questions on StackOverflow and elsewhere before you ask me or tell me "it can't be done". This is your chance to show that you can grow as a programmer.

ajcse1, 2018-01-08:
Done for all datasets.

closed 2018-01-08 by ajcse1