jaws/jaws #27: Add solar zenith angle

czender, 2018-01-08:
Please add solar zenith angle as a derived variable to all datasets. Derive this from the GMT using (hopefully) a pre-existing Python module, or ask me or Wenshan how to compute it.

ajcse1, 2018-01-26:
Calculated sza for all the stations. Need to generalize the code

ajcse1, 2018-01-26:
@wenshanw Can you please check whether the results of this formula are same as your calculations?

wenshanw, 2018-01-28:
I checked 24 hours. The largest difference is 0.7 deg. 
Would you please point me to the formula you use? @ajcse1 

(N)   Station  year mth day hour SZA_NCL SZA_python Difference
(0)	KAN_B 2011  4 13 0 96.0083  96.04462044964082  -0.03628915081269213
(1)	KAN_B 2011  4 13 1 90.4353  91.17425716189416  -0.7389467981246298
(2)	KAN_B 2011  4 13 2 85.4166  85.67734393099627  -0.2607713601954913
(3)	KAN_B 2011  4 13 3 79.7018  79.87603840142148  -0.1742027690972634
(4)	KAN_B 2011  4 13 4 73.9585  74.10371616002382  -0.1452582132464784
(5)	KAN_B 2011  4 13 5 68.5791  68.71185293040639  -0.1327284797228003
(6)	KAN_B 2011  4 13 6 63.9475  64.07434593832087  -0.1268933931548588
(7)	KAN_B 2011  4 13 7 60.45  60.57394645745858  -0.1239609533081847
(8)	KAN_B 2011  4 13 8 58.4333  58.55524528792849  -0.1219261961316107
(9)	KAN_B 2011  4 13 9 58.1256  58.24542214853581  -0.1198079822760434
(10)	KAN_B 2011  4 13 10 59.5628  59.6804939048347  -0.1176772849616512
(11)	KAN_B 2011  4 13 11 62.5752  62.69183510848241  -0.1166092784042831
(12)	KAN_B 2011  4 13 12 66.846  66.96446083082988  -0.1184830476267535
(13)	KAN_B 2011  4 13 13 71.9943  72.12055115167347  -0.1262960857554987
(14)	KAN_B 2011  4 13 14 77.6328  77.77932453668953  -0.1465730718457792
(15)	KAN_B 2011  4 13 15 83.379  83.5807509034959  -0.2017988771287094
(16)	KAN_B 2011  4 13 16 88.7438  89.18557379325536  -0.4417841204037956
(17)	KAN_B 2011  4 13 17 94.0567  94.2680508123387  -0.21134915218245
(18)	KAN_B 2011  4 13 18 98.517  98.51401048177406  0.003003068030622558
(19)	KAN_B 2011  4 13 19 101.621  101.6328362099309  -0.01223074118088618
(20)	KAN_B 2011  4 13 20 103.362  103.3865974244046  -0.02471998299830602
(21)	KAN_B 2011  4 13 21 103.599  103.6295795773468  -0.03078654756163246
(22)	KAN_B 2011  4 13 22 102.31  102.340133582111  -0.03008261775552512
(23)	KAN_B 2011  4 13 23 99.6031  99.62582688145889  -0.0227461319471729

ajcse1, 2018-01-28:
[https://github.com/jaws/jaws/source/sunposition.py](url) In this module at line #557, the function sunpos() is used to calculation sza.

closed 2018-02-14 by ajcse1

wenshanw, 2018-02-14:
sorry, Ajay, I haven't check your script yet. Will do this week!

Best,
Wenshan
------------------------------------------------------------------------------
Graduate Student Researcher
Earth System Science
University of California, Irvine
https://sites.google.com/a/uci.edu/wenshanwang/