jaws/jaws #28: Coordinates latitude, longitude, time must contain meaningful values

czender, 2018-01-09:
In particular, coordinates should never be empty, and time should increase monotonically. This shows that latitude and longitude are empty and time does not change:

```
netcdf promice_20160501 {
  dimensions:
    time = UNLIMITED ; // (24 currently)

  variables:
    float latitude(time) ;
      latitude:units = "ddmm" ;
      latitude:long_name = "Latitude GPS" ;
      latitude:standard_name = "latitude" ;

    float longitude(time) ;
      longitude:units = "ddmm" ;
      longitude:long_name = "Longitude GPS" ;
      longitude:standard_name = "longitude" ;

    int time(time) ;
      time:units = "seconds since 2007-01-01 00:00:00" ;
      time:long_name = "time of measurement" ;
      time:standard_name = "time" ;
      time:calendar = "standard" ;

  data:
    latitude = -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999 ; 

    longitude = -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999 ; 

    time = 294451200, 294451200, 294451200, 294451200, 294451200, 294451200, 294451200, 294451200, 294451200, 294451200, 294451200, 294451200, 294451200, 294451200, 294451200, 294451200, 294451200, 294451200, 294451200, 294451200, 294451200, 294451200, 294451200, 294451200 ; 

```

The fix for time is, well, to make it accurate. The changes to Lon & Lat are more subtle. Rename the existing longitude(time) and latitude(time) variables to longitude_GPS(time) and latitude_GPS(time). Only create these variables if the station has GPS.  Add new dimensions of size 1 (i.e., degenerate dimensions) named longitude and latitude. Add new coordinate variables longitude(longitude) and latitude(latitude) that store the time-invariant nominal latitude and longitude of the station. longitude and latitude should always exist for every station. Use CF standard units for latitude and longitude (degrees_north and degrees_east) information. Abandon "ddmm" units.

ajcse1, 2018-01-12:
For all datasets:
1. Corrected the time variable calculation. 
2. Added new coordinate variables longitude(longitude) and latitude(latitude)
3. Rename the existing longitude(time) and latitude(time) variables to longitude_GPS(time) and latitude_GPS(time). Abandoned the ddmm units

czender, 2018-01-12:
latitude* and longitude* must have meaningful values. 
```
zender@aerosol:~$ ncks -C -H --cal -v latitude.?,longitude.? ~/promice_20160501.nc 
netcdf promice_20160501 {
  dimensions:
    latitude = 1 ;
    longitude = 1 ;
    time = UNLIMITED ; // (24 currently)

  variables:
    float latitude(latitude) ;

    float latitude_GPS(time) ;

    float longitude(longitude) ;

    float longitude_GPS(time) ;

  data:
    latitude = 9.96921e+36 ; 

    latitude_GPS = -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999 ; 

    longitude = 9.96921e+36 ; 

    longitude_GPS = -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999 ; 

} // group /

```

czender, 2018-01-12:
Also I decided I made a poor decision in requesting latitude and longitude dimensions of size 1. Instead we should make latitude and longitude scalar coordinate dimensions:  http://cfconventions.org/Data/cf-conventions/cf-conventions-1.7/build/ch05s07.html
In other words, make latitude and longitude scalars, remove the latitude and longitude dimensions completly, and add the "coordinates" attribute with values "latitude longitude" to all applicable variables (most if not all non-coordinate variables). Then we will have, e.g.,
 ```
 float air_temperature(time) ;
      air_temperature:units = "kelvin" ;
      air_temperature:long_name = "Air Temperature" ;
      air_temperature:standard_name = "air_temperature" ;
     air_temperature:coordinates = "longitude latitude" ;
```

ajcse1, 2018-01-18:
Made  latitude and longitude scalar coordinate dimensions and added the "coordinates" attribute with values "latitude longitude" to all applicable variables for all the datasets.

closed 2018-01-22 by ajcse1