jaws/jaws #30: Add cell_methods attribute

czender, 2018-01-12:
Please add "cell_methods" attribute to most if not all geophysical variables that have a "time" dimension:

http://cfconventions.org/cf-conventions/cf-conventions.html#cell-methods

Do not add to time_bounds, that is the only exception I know of. So most variables will then look like this:
 ```
 float air_temperature(time) ;
      air_temperature:units = "kelvin" ;
      air_temperature:long_name = "Air Temperature" ;
      air_temperature:standard_name = "air_temperature" ;
      air_temperature:cell_methods = "time: mean" ;
```

czender, 2018-01-12:
If we ever get _instantaneous_ rather than time-mean AWS data, then its `cell_methods` should be `time: point`.

ajcse1, 2018-01-19:
Added the attribute for all datasets

closed 2018-01-22 by ajcse1