jaws/jaws #31: Station names: Add or lookup

czender, 2018-01-12:
The station name is currently unsupported. Sometimes is not directly present in the input ASCII data. In these cases, try to derive it from a JAWS-wide lookup table of (lat,lon,network_type)->Station name. In other words, JAWS needs an internal database of station location, name, network and possibly other ancillary information (such as maintenance reports) that is not present in the L2 input data. Users should also be able to set the value of "station_name" to anything they want, overriding the internal database, by specifying the command line option --station_name=foo during the translation phase.

ajcse1, 2018-01-18:
Added the station_name variable for all datasets. '--station_name' / '-s' provided as options to override the default value.

czender, 2018-01-19:
Code like this
		`station_name[0:6] = ['N','A','S','A','-','U']`
is not concise and does not employ Python's power. Learn and use a better method, something closer to
`station_name="NASA-U"`
The station_name code needs to be improved to use a table where the name is the key and the values are the lat, lon, and long_name of the station. Then you use a hash to retrieve that information.

ajcse1, 2018-01-24:
Created dictionary that stores  lat, lon, and long_name of the station for all the stations of all networks. station_name code uses list comprehension to store in better way.

closed 2018-02-01 by ajcse1