jaws/jaws #32: Add _FillValue

czender, 2018-01-12:
Add a [_FillValue](http://cfconventions.org/cf-conventions/cf-conventions.html#missing-data) attribute to any variable that has missing data. Missing data are usually indicated by -999 in PROMICE, for example. When complete, the output will look like
```
 variables:
    float air_temperature(time) ;
      air_temperature:units = "kelvin" ;
      air_temperature:long_name = "Air Temperature" ;
      air_temperature:standard_name = "air_temperature" ;
      air_temperature:_FillValue = -999f ;

  data:
    air_temperature = -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, -999, 253.42, 255.23, 256.31, 256.15, 254.89, 253.45, 250.65, 248.86, 246.43, 244.94 ; 

```

czender, 2018-01-16:
No _FillValue necessary for variables that always have valid data, e.g., lat, lon, time, station_name

ajcse1, 2018-01-17:
Added _FillValue for all necessary attributes.

closed 2018-01-22 by ajcse1

laffinm, 2019-06-27:
Add _FillValue for all hours for data that are 3-hourly temporal scale. This will ensure an hourly dataset with a value for all hours.