jaws/jaws #33: Fix canonical ordering

czender, 2018-01-12:
JAWS does not work with NCO canonical ordering of input/output filenames on the command line. Canonical ordering is: command options input output
```
zender@aerosol:~$ python ~/jaws/source/jaws.py ~/jaws/sample_data/promice_20160501.txt ~/promice_20160501.nc
usage: jaws.py [-h] [-o OUTPUT] [-3] [-4] [-5] [-6] [-7] input
jaws.py: error: unrecognized arguments: /Users/zender/promice_20160501.nc
```
This should be equivalent to 
```
zender@aerosol:~$ python ~/jaws/source/jaws.py -o ~/promice_20160501.nc ~/jaws/sample_data/promice_20160501.txt
converting data...
calculating time variable...
The file /Users/zender/jaws/sample_data/promice_20160501.txt is converted into netCDF format.
```
I think maybe this used to work and was broken by a recent change. This points to the need to start developing a unit test framework so that new features do not unwittingly break old ones.

ajcse1, 2018-01-12:
It is working correctly. I think you missed to give -o before the output filename in first command.

czender, 2018-01-12:
You have not read, or do not understand, canonical ordering: the last two positional arguments after all options correspond to the input file and the output file. No -o switch is necessary with canonical ordering. The "-o outfile" switch is optional, and, if used, must appear before the input filename. Re-read my messages on this. Copy what NCO does. Make the example above work.

ajcse1, 2018-01-17:
Also add -i option before the input file name.

ajcse1, 2018-02-01:
Fixed canonical ordering

closed 2018-02-02 by ajcse1