jaws/jaws #38: Functionalize common code

czender, 2018-01-19:
The three core routines repeat a lot of code that should be shared a single function. Functionalize (i.e., turn into functions/subroutines) code that is called more than once, so that there is only one copy of it. Repeating code is poor practice and leads to errors.  

This is repeated and should be a function:
```
	# NC file setup
	op_file = str((os.path.basename(args.input)).split('.')[0])+'.nc'
	
	if args.output:
		op_file = str(args.output)

	if args.format3 == 1:
		root_grp = Dataset(op_file, 'w', format='NETCDF3_CLASSIC')
	elif args.format4 == 1:
		root_grp = Dataset(op_file, 'w', format='NETCDF4')
	elif args.format5 == 1:
		root_grp = Dataset(op_file, 'w', format='NETCDF3_64BIT_DATA')
	elif args.format6 == 1:
		root_grp = Dataset(op_file, 'w', format='NETCDF3_64BIT_OFFSET')
	elif args.format7 == 1:
		root_grp = Dataset(op_file, 'w', format='NETCDF4_CLASSIC')
	else:
		root_grp = Dataset(op_file, 'w', format='NETCDF4')

```
Same with this
```
	# dimension
        stn_nm_lng_max=25
	root_grp.createDimension('time', None)
	root_grp.createDimension('nbnd', 2)
	root_grp.createDimension('station', 1)
	root_grp.createDimension('stn_nm_lng_max', stn_nm_lng_max)

	# variables
	station_name = root_grp.createVariable('station_name', 'S1', ('stn_nm_lng_max',))
	latitude = root_grp.createVariable('latitude', 'f4')
	longitude = root_grp.createVariable('longitude', 'f4')
	time = root_grp.createVariable('time', 'i4', ('time',))
	time_bounds = root_grp.createVariable('time_bounds', 'i4', ('time','nbnd'))

```
In general, the codebase should call common functions, and the network-specific modules should only contain code specific to that network.

ajcse1, 2018-01-22:
Sure, I was thinking about it. I will do that.

ajcse1, 2018-01-22:
Do also for analysis scripts

ajcse1, 2018-02-24:
Common coed has been functionalized

closed 2018-03-28 by ajcse1