jaws/jaws #39: Analysis scripts to be called from jaws.py

ajcse1, 2018-01-22:


czender, 2018-01-26:
I'm not sure I agree with this. It introduces dependencies of jaws.py on graphics packages. In any case, please robustly test any scripts for breakage before posting them on the homepage:
```
zender@aerosol:~/anaconda/bin$ cd ~/jaws/analysis;python analysis.py ~/GCNET_Summit_20140601.nc temperature_tc_1 diurnal -y 2002 -m 5
Traceback (most recent call last):
  File "analysis.py", line 21, in <module>
    ds = xarray.open_dataset(args.input)
  File "/Users/zender/anaconda/lib/python2.7/site-packages/xarray/backends/api.py", line 305, in open_dataset
    return maybe_decode_store(store, lock)
  File "/Users/zender/anaconda/lib/python2.7/site-packages/xarray/backends/api.py", line 225, in maybe_decode_store
    drop_variables=drop_variables)
  File "/Users/zender/anaconda/lib/python2.7/site-packages/xarray/conventions.py", line 1155, in decode_cf
    decode_coords, drop_variables=drop_variables)
  File "/Users/zender/anaconda/lib/python2.7/site-packages/xarray/conventions.py", line 1088, in decode_cf_variables
    stack_char_dim=stack_char_dim)
  File "/Users/zender/anaconda/lib/python2.7/site-packages/xarray/conventions.py", line 1019, in decode_cf_variable
    if 'since' in attributes['units']:
TypeError: argument of type 'numpy.int64' is not iterable

```

czender, 2018-01-26:
I guess the point of this is to provide more intuitive error messages. In this case, the file was for a day but the analysis was for a month of a different year.

ajcse1, 2018-01-26:
Alright, will provide more intuitive error messages.

ajcse1, 2018-02-02:
Resolved

ajcse1, 2018-02-23:
Printing meaningful error messages now.

closed 2018-02-23 by ajcse1