jaws/jaws #41: Different 'time' for time-averaged and instantaneous variables?

wenshanw, 2018-01-28:
For time-averaged variables such as radiation, the actual timestamp should be "0.5, 1.5, 2.5 ...".
For instantaneous variables such as elevation, it is "0, 1, 2 ...".
Should we keep two sets of 'time' in the file?
Or we can notify users in the global attributes?

czender, 2018-01-29:
Let's discuss in group meeting. The `cell_methods` descriptor for each variable should indicate the type of variable with `cell_methods=time: point` or `cell_methods=time: mean`. This can help with disambiguation, e.g., `time` can be `[0, 1, 2...]`, time_bounds=[[0,1],[1,2],[2,3]] then we only need one time variable and cell_methods in each variable informs user which time value is intended.

czender, 2018-01-29:
Please add global attribute named "`time_convention`" that explains that time: point variables match the time coordinate values exactly whereas time: mean variables are valid for the mean time withing the time_bounds variable. For example, in PROMICE datasets sensor_height is measure once per hour at the time stored in the "time" coordinate, and temperature is continuously measured and then hourly-mean values are stored for each period contained in the time_bounds variable.

ajcse1, 2018-02-01:
Added for all datasets

closed 2018-02-13 by ajcse1