jaws/jaws #48: Refactor code to improve throughput

czender, 2018-02-01:
There is no good reason why it should take 10s of minute to hourse to convert a GCNet file. The code needs refactoring. Craft a well-posed question about how to speed it up, post it in the appropriate location (StackOverflow, xarray help, whatever), and send me a link so I can follow the responses.

ajcse1, 2018-02-05:
Currently implementing a different approach to read and write data. Will post once I get the running time of new code.

czender, 2018-02-06:
Please base the approach on `xarray` capabilities and point me to a Question/Answer that verifies the approach is an  `xarray` best practice. For reasons of future collaboration, JAWS will employ xarray techniques where possible.

ajcse1, 2018-02-07:
Sure, I am now working on basing the approach on xarray. This is the link [https://stackoverflow.com/a/28914767/9326176](url) for one of the answers on SO that I fund useful.

ajcse1, 2018-02-08:
Questions asked:
[https://stackoverflow.com/questions/48679362/](url)
[https://stackoverflow.com/questions/48681864](url)

czender, 2018-02-08:
These are great suggestions, and will improve the code quality and speed tremendously. One day you will give such answers. Until then the experience of other Python/xarray programmers is a gift to utilize each time you face a new problem where you suspect there is a simpler/faster answer.

czender, 2018-02-13:
This sounds great but because of all the bugs I cannot test it.

ajcse1, 2018-02-15:
PROMICE calculation has also been refactored and now takes only few minutes for processing complete file.

closed 2018-02-24 by ajcse1