jaws/jaws #49: Test Modules

ajcse1, 2018-02-05:
Implement unit tests for all scripts

czender, 2018-02-13:
This is an important task. Has it become clear that JAWS may work for you but not others? At a minimum we need automatic verification that a default call each of the networks works. If you make JAWS a conda-installable package, then you can have TravisCI and AppVeyor automatically test it on a build-matrix of machines after each commit, or each tagged commit (as NCO does). I suggest you look to StackOverflow or GitHub help sites on best practices to turn a Python project into a Conda package and invoke automatic testing.

czender, 2018-03-12:
Please update and close once Travis is guaranteed working with no false positives

czender, 2018-03-12:
Please add following tests:
One where as many options as possible are perfomed in one invocation
One that tests compression

ajcse1, 2018-03-29:
Added both of the above mentioned tests and checked for false positives.

closed 2018-03-29 by ajcse1