jaws/jaws #52: Breakage due to timezone option

czender, 2018-02-13:
```
zender@firn:~/anaconda/bin$ python ~/jaws/source/jaws.py ~sainiak/data/jaws/gcnet/raw/gcnet_summit_19960512_20170524.dat_Req1666 ~/data/jaws/gcnet_summit_19960512_20170524.nc
Traceback (most recent call last):
  File "/Users/zender/jaws/source/jaws.py", line 311, in <module>
    Main()
  File "/Users/zender/jaws/source/jaws.py", line 20, in Main
    parser.add_argument("-t","--timezone", help = "Change the timezone, default is UTC", default='UTC', type='str')
  File "/Users/zender/anaconda/lib/python2.7/argparse.py", line 1299, in add_argument
    raise ValueError('%r is not callable' % (type_func,))
ValueError: 'str' is not callable
```

czender, 2018-02-13:
```
zender@skyglow:~/nco$ python ~/jaws/source/jaws.py ~sainiak/data/jaws/gcnet/raw/gcnet_summit_19960512_20170524.dat_Req1666 ~/data/jaws/gcnet_summit_19960512_20170524.nc
Traceback (most recent call last):
  File "/home/zender/jaws/source/jaws.py", line 3, in <module>
    from netCDF4 import Dataset
  File "/home/zender/anaconda/lib/python2.7/site-packages/netCDF4/__init__.py", line 3, in <module>
    from ._netCDF4 import *
ImportError: /usr/lib64/libk5crypto.so.3: symbol k5_once, version krb5support_0_MIT not defined in file libkrb5support.so.0 with link time reference
zender@skyglow:~/nco$ ls ~sainiak/data/jaws/gcnet/raw/gcnet_summit_19960512_20170524.dat_Req1666
ls: cannot access '~sainiak/data/jaws/gcnet/raw/gcnet_summit_19960512_20170524.dat_Req1666': No such file or directory

```

czender, 2018-02-13:
here the input file does exist but same error as original post, so this is a bug and I will open another issue for it.

```
zender@grele:~/anaconda/bin$ python ~/jaws/source/jaws.py ~sainiak/data/jaws/gcnet/raw/gcnet_summit_19960512_20170524.dat_Req1666 ~/data/jaws/gcnet/gcnet_summit_19960512_20170524.nc
Traceback (most recent call last):
  File "/home/zender/jaws/source/jaws.py", line 311, in <module>
    Main()
  File "/home/zender/jaws/source/jaws.py", line 20, in Main
    parser.add_argument("-t","--timezone", help = "Change the timezone, default is UTC", default='UTC', type='str')
  File "/home/zender/anaconda/lib/python2.7/argparse.py", line 1299, in add_argument
    raise ValueError('%r is not callable' % (type_func,))
ValueError: 'str' is not callable
zender@grele:~/anaconda/bin$ ls -l ~sainiak/data/jaws/gcnet/raw/gcnet_summit_19960512_20170524.dat_Req1666
-rw-rw-r-- 1 sainiak sainiak 62140412 Jan 30 22:09 /home/sainiak/data/jaws/gcnet/raw/gcnet_summit_19960512_20170524.dat_Req1666

```

ajcse1, 2018-02-13:
_jaws_ throws the following error if nor input file is provided:

`Error: You failed to provide input file!`

czender, 2018-02-13:
Please re-read the issue. I was asking that when the input filename is provided, but that file does not exist (as in the failed examples) a helpful message is printed. When required command-line arguments are omitted or misused, then JAWS should print the correct usage, as NCO does. Whenever you wonder what JAWS should do, ask yourself if there is an NCO analog, and make JAWS do that:
```
zender@firn:~/jaws$ python ~/jaws/source/jaws.py
Error: You failed to provide input file!
zender@firn:~/jaws$ ncks
ncks: ERROR received 0 filenames; need at least one
ncks Command line options cheatsheet (full details at http://nco.sf.net/nco.html#ncks):
ncks [-3] [-4] [-5] [-6] [-7] [-A] [-a] [-b fl_bnr] [--bfr byt] [-C] [-c] [--cal] [--cdl] [--cnk_byt byt] [--cnk_csh byt] [--cnk_dmn nm,lmn] [--cnk_map map] [--cnk_min byt] [--cnk_plc plc] [--cnk_scl sz] [-D nco_dbg_lvl] [-d ...] [--dt_fmt fmt] [-F] [--fix_rec_dmn dim] [--fl_fmt fmt] [-G grp:lvl] [-g ...] [--glb ...] [--grp_xtr_var_xcl] [-H] [-h] [--hdn] [--hdr_pad nbr] [--jsn] [--jsn_fmt lvl] [-L lvl] [-l path] [-M] [-m] [--map map.nc] [--md5_dgs] [--md5_wrt] [--mk_rec_dmn dim] [--msa] [--no_blank] [--no_cll_msr] [--no_frm_trm] [--no_tmp_fl] [-O] [-o out.nc] [-P] [-p path] [--ppc ...] [-Q] [-q] [-R] [-r] [--rad] [--rgr] [--ram_all] [--rnr=wgt] [-s format] [-t thr_nbr] [-u] [--unn] [-V] [-v ...] [-X box] [-x] [--xml] [--xml_no_loc] [--xml_spr_chr sng] [--xml_spr_nmr sng] [--xtn_var ...] in.nc [[out.nc]]

-3, --3, --fl_fmt=classic	Output file in netCDF3 CLASSIC (32-bit offset) storage format
-4, --4, --netcdf4		Output file in netCDF4 (HDF5) storage format
-5, --5, --fl_fmt=64bit_data	Output file in netCDF3 64-bit data (i.e., CDF5, PnetCDF) storage format
-6, --6, --64, --fl_fmt=64bit_offset	Output file in netCDF3 64-bit offset storage format
-7, --7, --fl_fmt=netcdf4_classic	Output file in netCDF4 CLASSIC format (3+4=7)
-A, --apn, --append	Append to existing output file, if any
-a, --abc, --alphabetize	Disable alphabetization of extracted variables
-b, --fl_bnr, --binary-file fl_bnr	Unformatted binary file to write
    --bfr_sz, --buffer_size sz	Buffer size to open files with
-C, --nocoords		Associated coordinate variables should not be processed
-c, --crd, --coords	Coordinate variables will all be processed
    --cal,--cln	Print UDUnits-compatible dates/times in human-legible calendar format
    --cdl		Print CDL (netCDF lingua franca used by ncdump/ncgen)
    --cnk_byt, --chunk_byte sz_byt	Chunksize in bytes
    --cnk_csh, --chunk_cache sz_byt	Chunk cache size in bytes
    --cnk_dmn, --chunk_dimension nm,sz_lmn	Chunksize of dimension nm (in elements not bytes)
    --cnk_map, --chunk_map map		Chunking map [dmn,lfp,nc4,nco,prd,rd1,rew,scl,xpl,xst]
    --cnk_min, --chunk_min sz_byt	Minimum size [B] of variable to chunk
    --cnk_plc, --chunk_policy plc	Chunking policy [all,g2d,g3d,xpl,xst,uck]
    --cnk_scl, --chunk_scalar sz_lmn	Chunksize scalar (in elements not bytes) (for all dimensions)
-D, --dbg_lvl, --debug-level lvl	Debug-level is lvl
-d, --dmn, --dimension dim,[min][,[max]][,[stride]] Dimension's limits and stride in hyperslab
    --dt_fmt,--date_format	Print UDUnits-compatible dates/times in short, regular, or ISO8601 calendar format (fmt=1,2,3)
-F, --ftn, --fortran	Fortran indexing conventions (1-based) for I/O
    --fix_rec_dmn dim	Change dimension dim (or all) to fixed dimension in output file
    --fl_fmt, --file_format fmt	File format for output [classic,64bit_offset,64bit_data,netcdf4,netcdf4_classic]
-G, --gpe [grp_nm][:[lvl]]	Group Path Editing path, levels to replace
-g, --grp grp1[,grp2[...]] Group(s) to process (regular expressions supported)
    --glb_att_add nm=val	Global attribute to add
-H, --data, --hieronymus	Toggle printing data
-h, --hst, --history	Do not append to "history" global attribute
    --hdn, --hidden	Print hidden (aka special) attributes
    --hdr_pad, --header_pad	Pad output header with nbr bytes
    --jsn, --json	Print JSON (JavaScript Object Notation)
    --jsn_fmt lvl	Verbosity of JSON format [0 = least verbose, 2 = most pedantic, add 4 to remove brackets]
-L, --dfl_lvl, --deflate lvl	Lempel-Ziv deflation/compression (lvl=0..9) for netCDF4 output
-l, --lcl, --local path	Local storage path for remotely-retrieved files
-M, --Mtd, --Metadata	Toggle printing global metadata
-m, --mtd, --metadata	Toggle printing variable metadata
    --map, --rgr_map map.nc	File containing (ESMF- or SCRIP-format) weights to regrid input to output grid
    --md5_dgs, --md5_digest	Perform MD5 digests
    --md5_wrt, --md5_write	Write MD5 digests as attributes
    --mk_rec_dmn dim	Define dim as record dimension in output file
    --no_blank		Print numeric missing values instead of blanks (underscores)
    --no_cll_msr	Do not extract cell_measures variables
    --no_frm_trm	Do not extract formula_terms variables
    --no_tmp_fl		Do not write output to temporary file
-O, --ovr, --overwrite	Overwrite existing output file, if any
-o, --output, --fl_out out.nc	Output file name (or use last positional argument)
-P, --prn, --print	Print data, metadata, and units. Abbreviation for -C -H -M -m -u.
-p, --pth, --path path	Path prefix for all input filenames
    --ppc v1[,v2[,...]]=prc	Precision-Preserving Compression: Number of total or decimal significant digits
-Q, --quiet		Quiet printing of dimension indices and coordinate values
-q, --quench		Quench (turn-off) all printing to screen
-R, --rtn, --retain	Retain remotely-retrieved files after use
[... cut ...]
3. User Guide:   http://nco.sf.net#RTFM
4. Manual pages: 'man ncks', 'man nco', ...
5. Homepage:     http://nco.sf.net
6. FAQ:          http://nco.sf.net#FAQ
7. Help Forum:   http://sf.net/p/nco/discussion/9830
8. Publications: http://nco.sf.net#pub
Post questions, suggestions, patches at http://sf.net/projects/nco
zender@firn:~/jaws$ 
```
JAWS has an acceptable error message for when the input file does not exist. Please revert your and make JAWS print a usage summary like NCO when the command line arguments are missing or misused.

ajcse1, 2018-02-16:
Now, printing helpful error messages

closed 2018-02-22 by ajcse1