jaws/jaws #54: Use netCDF-default _FillValues

czender, 2018-02-13:
Instead of 999 or something, please make JAWS use the default netCDF _FillValue for each type of variable. And never use NaN:
zender@firn:~/data/jaws$ ncks -C -v julian_decimal_time -m gcnet_summit_19960512_20170524.nc
```
netcdf gcnet_summit_19960512_20170524 {
  dimensions:
    time = UNLIMITED ; // (184369 currently)

  variables:
    double julian_decimal_time(time) ;
      julian_decimal_time:_FillValue = NaN ;
      julian_decimal_time:units = "decimal time" ;
      julian_decimal_time:long_name = "Julian Decimal Time" ;
      julian_decimal_time:note = "For each year, time starts at 1.0000 and ends at 365.9999." ;
} // group /
```

ajcse1, 2018-02-15:
Using default netCDF _FIllValue for all variables in JAWS. Also, NaN was assigned automatically by xarray if we don't set any value. Now, set the _FillValue as _False_ for all the variables that are always present.

wenshanw, 2018-11-19:
Don't forget 'Latitude', 'longitude', and 'sza'~

ajcse1, 2018-11-19:
'Latitude', 'longitude', and 'sza' are always present, that's why _FillValue has been set as _False_ for them.

ajcse1, 2018-12-07:
Fixed in v0.6.3. Added 'derived_vars', 'no_drv_tm_vars' and 'rigb_vars' to the encoding list

closed 2018-12-07 by ajcse1