jaws/jaws #55: Create Conda Package

czender, 2018-02-21:
Make JAWS a conda package on the conda-forge channel

ajcse1, 2018-03-29:
The package can be installed using :

`conda install -c jaws jaws`

Earlier I was getting error `extern "Python": function Cryptography_locking_cb() called, but @ffi.def_extern() was not called in the current subinterpreter. Returning 0.`

This has been resolved now and package is installing now for Windows without any error.

For Linux/Mac, getting error:

`The path specified in the package manifest cannot be found`

Following are the links, where users faced similar problems: 
https://github.com/Microsoft/AzureNotebooks/issues/88
Comment section of http://www.codesofinterest.com/2017/05/opencv-32-and-dlib-194-available-conda-forge.html
https://github.com/ContinuumIO/anaconda-issues/issues/8912

The solution posted in these and other links didn't work for our case. Still working on it.

czender, 2018-03-29:
The desired installation command is
`conda install -c conda-forge jaws`.
We want the channel to be `conda-forge` and the package to be `jaws`.
The channel is not `jaws`.
It is an exact parallel, as usual, to NCO being installed with `conda install -c conda-forge nco`.
Now, does that clarification help solve any issues?

czender, 2018-03-29:
JAWS appears not be registered on the conda-forge channel:

```
zender@aerosol:~/anaconda/bin$ conda install -c conda-forge jaws
Fetching package metadata .............

PackageNotFoundError: Packages missing in current channels:
            
  - jaws

We have searched for the packages in the following channels:
            
  - https://conda.anaconda.org/conda-forge/osx-64
  - https://conda.anaconda.org/conda-forge/noarch
  - https://repo.continuum.io/pkgs/main/osx-64
  - https://repo.continuum.io/pkgs/main/noarch
  - https://repo.continuum.io/pkgs/free/osx-64
  - https://repo.continuum.io/pkgs/free/noarch
  - https://repo.continuum.io/pkgs/r/osx-64
  - https://repo.continuum.io/pkgs/r/noarch
  - https://repo.continuum.io/pkgs/pro/osx-64
  - https://repo.continuum.io/pkgs/pro/noarch

```

ajcse1, 2018-03-29:
I was waiting for clean installation of _jaws_ before putting it on conda-forge channel. 
'jaws' channel was just created for temporary testing of installation.

ajcse1, 2018-03-30:
_jaws_ can now be installed with:

`conda install -c jaws jaws`

for any platform and both Python versions.

There was some issue with new version of anaconda, had to downgrade to anaconda-4.4 and then build the package.

Also, submitted a PR on conda-forge to make it available through that channel.

ajcse1, 2018-04-06:
Made changes to recipe as requested by conda-forge team. Waiting for their update.

czender, 2018-04-19:
Great!

closed 2018-04-24 by ajcse1