jaws/jaws #60: Unsynchronized time and/or timestamp variable in AAWS

czender, 2018-02-23:
`time` and `timestamp` are not syncrhonized, why?
I suspect the issue is systemic, at least for AAWS.
Please verify whether similar issues affect GCNET and PROMICE.
This underscores the importance of spot-checking the files.
Problems like this are only detectable by reading the files occasionally.
`time` is the most important variable in a file and its representation must be impeccable.

```
zender@firn:~/data/jaws/aaws/nc$ ncks -d time,0 --cal -v timestamp aaws_elizabeth_20010101_20041231.nc
netcdf aaws_elizabeth_20010101_20041231 {
  dimensions:
    nbnd = 2 ;
    string20 = 20 ;
    time = UNLIMITED ; // (1 currently)

  variables:
    double time(time) ;
      time:units = "seconds since 1970-01-01 00:00:00" ;
      time:long_name = "time of measurement" ;
      time:standard_name = "time" ;
      time:bounds = "time_bounds" ;
      time:calendar = "noleap" ;

    double time_bounds(time,nbnd) ;

    char timestamp(time,string20) ;

  data:
    time = "2001-01-09" ;

    time_bounds = 
    "2001-01-08 23:00:00", "2001-01-09" ;

    timestamp = 
    "2001-01-01T00:00:00Z" ;

} // group /

```

ajcse1, 2018-02-23:
Corrected the error, it was because calendar was set as 'noleap', now changed to 'leap'.

czender, 2018-02-23:
Good. Please verify that the dates in the filename match the time variable in the file and the timestamp variable, if any, for all files in all networks before you close this issue. I verified that the "leap" fix appears to have fixed the PROMICE files.

ajcse1, 2018-02-24:
Checked for all networks

closed 2018-02-24 by ajcse1