jaws/jaws #62: Debugging levels

ajcse1, 2018-02-26:
Change print messages to verbose debugging levels. e.g:
Level 0: No messages
Level 1: Converted i/p file path to o/p file path in elapsed time.
Level 2: Filenames
and so on

czender, 2018-02-26:
nco/src/nco/nco.h contains this list:
```
  enum nco_dbg_typ_enm{ /* [enm] Debugging levels */
    /* List in increasing levels of verbosity */
    nco_dbg_quiet, /* 0 [enm] Quiet all non-error messages */
    nco_dbg_std,   /* 1 [enm] Standard mode. Minimal, but some, messages */
    nco_dbg_fl,    /* 2 [enm] Filenames */
    nco_dbg_scl,   /* 3 [enm] Scalars, other per-file information */
    nco_dbg_grp,   /* 4 [enm] Groups, highest level per-file loop information */
    nco_dbg_var,   /* 5 [enm] Variables, highest level per-group loop information */
    nco_dbg_crr,   /* 6 [enm] Current task */
    nco_dbg_sbr,   /* 7 [enm] Subroutine names on entry and exit */
    nco_dbg_io,    /* 8 [enm] Subroutine I/O */
    nco_dbg_vec,   /* 9 [enm] Entire vectors */
    nco_dbg_vrb,   /* 10 [enm] Verbose, print everything possible */
    nco_dbg_old,   /* 11 [enm] Old debugging blocks not used anymore */
    nco_dbg_dev,   /* 12 [enm] NCO developer information; only useful for debugging */
    nco_dbg_sup,   /* 13 [enm] NCO extra developer information not usually used (debug memory leaks) */
    nco_dbg_nbr    /* 14 [enm] Number of debugging types (equals last enumerated value) */
  }; /* end nco_dbg_typ_enm */
```
Please replace nco_ by jaws_ and adopt that naming format in the code.
jaws_dbg_std should be elapsed time.
jaws_dbg_fl should add input/output filepaths.

ajcse1, 2018-03-06:
All levels added

closed 2018-03-06 by ajcse1