jaws/jaws #67: Capitalize values of long_name attribute

czender, 2018-03-14:
The value of the long_name attribute is the default axis label for plotting the numeric values. Therefore it should be capitalized, e.g., "Air Pressure" not "air pressure". This will need to be done by hand. Follow standard capitalization rules and leave prepositions lowercase. Ask if unsure.

ajcse1, 2018-03-16:
Made all changes

czender, 2018-03-30:
The present status slide for the ESTO review clearly shows some long_name attributes whose values are not capitalized and suitable for axis labels. Please revisit this issue and fix. Slides should reflect the best that JAWS can currently do.

ajcse1, 2018-03-30:
Sorry, used converted file from an older version of jaws for that slide. Changed that slide now. All long_name attributes for each network are capitalized.

closed 2018-03-30 by ajcse1