jaws/jaws #70: Add CMR Metadata

ajcse1, 2018-04-05:


czender, 2018-04-12:
Look [here ](http://dust.ess.uci.edu/tmp/005_newport_pier-2018.nc)

czender, 2018-04-12:
Not all the global metadata in that file is CMR, though some appears to be. Look for the "low-hanging fruit", the important attributes, and develop a list of what you think we should add to JAWS, and post it here and we will discuss. You can find contacts at ESIP in Tucson who will know more about CMR compliance.

ajcse1, 2018-04-23:
Important attributes:
1. data_max, valid_max
2. data_min, valid_min
3. axis for some variables
4. instrument
5. axis for lat and lon

ajcse1, 2018-05-02:
Global attributes:
1. Platform: ShortName, LongName, Type, Characteristics
2. Instrument
3. Science Keywords

closed 2019-09-25 by ajcse1