jaws/jaws #76: Update examples to use conda-installed JAWS

czender, 2018-04-19:
Please update the examples to assume use of the conda-forge-installed version of JAWS, since that will be the preferred way to install it. This means 
1. Change the web page to show exactly how to install JAWS into a fresh conda installation
2. Command-line invocation converted to use conda-JAWS
3. Alter internal documentation accordingly
4. Ensure examples work on Linux and Windows for yourself
5. @wenshanw please verify the documented method works for you, too

ajcse1, 2018-04-19:
Yes, working on that

ajcse1, 2018-04-23:
Updated instructions for installation and example.

Installation succesfull on all platforms Linux, Mac, Windows.

Command line invocation working on Linux and Mac. Working on fixing it for Windows.

ajcse1, 2018-04-27:
Command line invocation works on Windows as well

closed 2018-04-27 by ajcse1