jaws/jaws #81: Calling analysis.py

ajcse1, 2018-04-25:
What alias should be given to call analysis.py from command line?
I can think of "jaws-analysis". Any other suggestions?

czender, 2018-04-26:
I favor having only one entry point, jaws, and reframing the analysis as a subroutine that jaws.py calls. Instead of

`python analysis.py file.nc temperature_tc_1 diurnal -y 2002 -m 5
`
the API would be something like

```
jaws -a diurnal -v temperature_tc_1 -y 2002 -m 5 file.nc
jaws --anl=diurnal --var=temperature_tc_1 --anl_yr=2002 --anl_mth=5 file.nc
jaws --analysis=diurnal --variable=temperature_tc_1 --analysis_year=2002 --analysis_month=5 file.nc
```
When the option parser sees the a/anl/analysis option is invoked, then JAWS goes into analysis mode instead of translation mode. Can you make this work?

ajcse1, 2018-04-26:
Agree, that's a good approach. Let me try, I think so it can be done.

ajcse1, 2018-05-01:
Re-framed analysis.py as a subroutine that jaws.py calls. Above mentioned API now works on all platforms. Updated instructions on homepage.

closed 2018-05-01 by ajcse1