jaws/jaws #87: Pip package

ajcse1, 2018-05-01:
Should we make JAWS available through pip?

czender, 2018-05-01:
What level of effort (how many days) would this require?

ajcse1, 2018-05-01:
Approximately 2-4 days

czender, 2018-05-01:
Yes, please proceed.

ajcse1, 2018-05-02:
The package is ready to upload but 'jaws' name is already taken by someone on PyPI. That [package](https://pypi.org/project/jaws/) was last updated in 2012 and the project Github link exists no more. To claim the name, the official documentation states the following:

> How do I claim an abandoned or previously registered project name?
> There is currently no established process for performing this administrative task that is explicit and fair for all parties. However, one is currently in development per PEP 541.
> 
> PEP 541 has been accepted, and the project is creating a workflow which will be documented here.

Here is a [Github Issue](https://github.com/pypa/warehouse/issues/1506) keeping track of it.

Similar [request](https://github.com/pypa/pypi-legacy/issues/789) was opened by another user recently and they have asked to wait.

czender, 2018-05-02:
In the meantime can you use a different name for the PyPI package, e.g., jaws_aws? Or contact the current owner and ask him to give you ownership so you can completely replace the old package with JAWS?

ajcse1, 2018-05-02:
Requested transfer of ownership from current owner

ajcse1, 2018-05-03:
Andrey Popp transferred ownership. JAWS is now installable through `pip install jaws`. Checked for all platforms and updated instructions on homepage.

closed 2018-05-03 by ajcse1