jaws/jaws #95: Add --version option

czender, 2018-05-23:
Please add a --version option so that this command prints the version and date last modified of the invoked script:
```
zender@aerosol:~/anaconda/bin$ jaws --version
usage: jaws [-h] [-i FL_IN] [-o FL_OUT] [-3] [-4] [-5] [-6] [-7] [-d]
            [-s STN_NM] [-t TZ] [-f FLL_VAL_FLT] [-D DBG_LVL] [-L DFL_LVL]
            [-a ANL] [-v VAR] [-y ANL_YR] [-m ANL_MTH]
            [input_file] [output_file]
jaws: error: unrecognized arguments: --version

```

ajcse1, 2018-05-24:
Added option.

czender, 2018-06-28:
Please add the JAWS version in the produced file header, similar to NCO:

 ```
   :NCO = "netCDF Operators version 4.7.6-alpha01 (Homepage = http://nco.sf.net, Code = http://github.com/nco/nco)" ;

   :JAWS = "Justified Automatic Weather Station software version 4.7.6-alpha01 (Homepage = http://github.com/jaws/jaws)" ;

```

ajcse1, 2018-06-29:
Added

closed 2018-06-29 by ajcse1