jaws/jaws #106: Ignore whitespace in header when looking for column names like station ID

czender, 2018-06-28:
A user at GEUS was foiled when a new PROMICE dataset he tried would not work with JAWS. The problem was that the new header had a tab instead of a space in the header. Please make JAWS treat all whitespace as equal when processing files, so that tabs vs. spaces does not break things.

ajcse1, 2018-06-29:
When JAWS reads input file, we need to provide what delimiter is used to separate different columns. Right now, it is set to as single space. We can provide an option as '--dlm/--delimiter', using which user can provide any kind of delimiter(space, tab, comma, colon, etc.) that the file has. The default can be set as what we have for sample files for each network.

czender, 2018-06-29:
That solution is unlikely to work for users. The default delimiter should be a string (multiple characters) that matches (in the regular expression sense) any combination of whitespace.  In other words, use regular expression matching not exact character matches, to identify different columns.

ajcse1, 2018-06-30:
Agree. Working to implement this.

ajcse1, 2018-07-09:
Done

closed 2018-07-09 by ajcse1