jaws/jaws #115: Fix cython dependency

czender, 2018-07-16:
I had been running JAWS fine on my Mac until I upgraded to 0.4.7. Then the lack of cython broke it. Shouldn't JAWS automatically install all necessary packages?
```
zender@aerosol:~/anaconda/bin$ jaws -4 ${DATA}/jaws/promice/raw/promice_Kangerlussuaq-U_20090404_20170916.txt ~/promice_KanU.nc
Traceback (most recent call last):
  File "./jaws", line 6, in <module>
    from pkg_resources import load_entry_point
  File "/Users/zender/anaconda/lib/python2.7/site-packages/pkg_resources/__init__.py", line 3147, in <module>
    @_call_aside
  File "/Users/zender/anaconda/lib/python2.7/site-packages/pkg_resources/__init__.py", line 3131, in _call_aside
    f(*args, **kwargs)
  File "/Users/zender/anaconda/lib/python2.7/site-packages/pkg_resources/__init__.py", line 3160, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/Users/zender/anaconda/lib/python2.7/site-packages/pkg_resources/__init__.py", line 666, in _build_master
    ws.require(__requires__)
  File "/Users/zender/anaconda/lib/python2.7/site-packages/pkg_resources/__init__.py", line 984, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/Users/zender/anaconda/lib/python2.7/site-packages/pkg_resources/__init__.py", line 870, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'cython' distribution was not found and is required by cftime
zender@aerosol:~/anaconda/bin$ which jaws
.//jaws
zender@aerosol:~/anaconda/bin$ conda install cython
Solving environment: done

## Package Plan ##

  environment location: /Users/zender/anaconda

  added / updated specs: 
    - cython

The following packages will be downloaded:

    package                    |            build
    ---------------------------|-----------------
    cython-0.28.4              |   py27hfc679d8_0         3.3 MB  conda-forge

The following NEW packages will be INSTALLED:

    cython: 0.28.4-py27hfc679d8_0 conda-forge

Proceed ([y]/n)? y

Downloading and Extracting Packages
cython-0.28.4        |  3.3 MB | ################################################################### | 100% 
Preparing transaction: done
Verifying transaction: done
Executing transaction: done
zender@aerosol:~/anaconda/bin$ jaws -4 ${DATA}/jaws/promice/raw/promice_Kangerlussuaq-U_20090404_20170916.txt ~/promice_KanU.nc

```

ajcse1, 2018-07-30:
Fixed in version 0.4.8

closed 2018-07-30 by ajcse1

xylar, 2018-08-14:
@czender and @ajcse1, in case this issue ever crops up again, I think it will be solved by https://github.com/Unidata/cftime/pull/62.  I had the same issue come up in one of my repos.

ajcse1, 2018-08-14:
Thanks @xylar! For us it was fixed with a new version release, without any changes to the requirements/package dependencies. So, we weren't sure what caused this error and how it got fixed. Now, it makes sense why it happened. Thanks again for the reference.