jaws/jaws #116: Fix error with GCNet data obtained by Dave Reusch

czender, 2018-07-17:
http://dust.ess.uci.edu/tmp/01c.dat_Req1217

contains a GCNet dataset obtained by Dave Reusch of NMIT. JAWS 0.4.7 fails to convert it. Please first verify that you can reproduce the problem. Then try to fix JAWS to process it. Dave is here in Madison and it would be great to provide him with a quick fix.

ajcse1, 2018-07-17:
The file has different header than what we received from GCNet. It doesn't have the quality control variables i.e. 'QCl01-08', 'QCl09-16', 'QCl17-24', 'QCl25-27' as well as another variable 'Battery Voltage'. 

JAWS calculates individual quality control variables, using the above mentioned variables. Since it can't find these variables, hence it fails to convert.

czender, 2018-07-17:
Seems like the JAWS procedure should be modified to first query the presence of a given field/column before adding that variable to the netCDF file and expecting each row to contain it. More if/thens, in other words. This is quite common for backwards compatibility with datasets that evolve through time.

ajcse1, 2018-07-24:
Working on it.

dbreusch, 2018-07-25:
So I wrote out a whole comment to the effect that my files "are as delivered to me and I definitely haven't modified them", then had a dim memory of a part of the data request process where you get to pick just the variables you want (see http://cires1.colorado.edu/science/groups/steffen//gcnet/order/admin/station.php).  I'm going to assume I didn't think I needed the QC vars (possibly a mistake in hindsight...).  But I can imagine people picking lots of permutations so perhaps it's good I tried a file-that's-different-from-yours and uncovered this issue.

Still looking forward to trying this out, it's such a great idea!

ajcse1, 2018-07-25:
Yeah, agree people will not be using all variables. Thanks for pointing out the issue. It will be fixed soon.

ajcse1, 2018-07-30:
Fixed in version 0.4.8. Now, user can choose any combination of variables in input file.
@dbreusch Can you please update _JAWS_ and test it? 
It will take around 4-5 minutes to completely convert this file.

dbreusch, 2018-07-30:
Working on my files now, thanks!

One minor comment:  station_numbers can go up to 23 so the metadata isn't quite right.  Looks like you're using the comment in the data file for this metadata.  That's out-of-date so not your fault it's wrong :-)

Two "fyi" benchmarks:
- my largest file, 60 MB, converted in 5m17s
- my smallest file, 2.4 MB, converted in 0m13s

ajcse1, 2018-07-30:
> One minor comment: station_numbers can go up to 23 so the metadata isn't quite right. Looks like you're using the comment in the data file for this metadata. That's out-of-date so not your fault it's wrong :-)

Thanks for pointing out. I will update it

closed 2018-07-30 by ajcse1