jaws/jaws #133: RIGB Issue with Antarctic IMAU station

czender, 2018-12-19:
Copying this so thread goes on GitHub. Please post followups here.

Dear Ajay, dear Charlie,

Today I installed JAWS using pip, and it is operating as expected, with great netCDF-formatted output from the L2 data of one of our Antarctic weather stations at IMAU.

However, I have come across an issue when invoking RIGB. Without —rigb, the conversion works smoothly, but with the option —rigb turned on, I get the error shown in the screen print below. Am I missing any packages perhaps? Or is there another issue?

My operating system is Mac OS High Sierra 10.13.6, Python 3.5.2 -- Enthought, Inc. (x86_64)

In the screen shot, you see succesful application of jaws in the last line, but an error in the previous command with the —rigb option.

Thanks in advance for your help!

With best regards,
Peter

pkuipersmunneke, 2018-12-20:
Hi Charlie and Ajay,
Please find the input file pasted below.
[ant_aws17_final_year2011.txt](https://github.com/jaws/jaws/files/2698697/ant_aws17_final_year2011.txt)

pkuipersmunneke, 2018-12-20:
Interestingly, if I select only the uppermost 10 lines, I get

kuipersm@:~/Data/AWS/Antarctica/AWS17> jaws --rigb -4 -o ant_aws17_final_year2011_TEST.nc ant_aws17_goodangles.txt
Traceback (most recent call last):
  File "/Users/kuipersm/Library/Enthought/Canopy/edm/envs/User/lib/python3.5/site-packages/xarray/core/dataset.py", line 780, in _construct_dataarray
    variable = self._variables[name]
KeyError: 'tilt_direction'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/kuipersm/Library/Enthought/Canopy/edm/envs/User/bin/jaws", line 11, in <module>
    sys.exit(start())
  File "/Users/kuipersm/Library/Enthought/Canopy/edm/envs/User/lib/python3.5/site-packages/jaws/jaws.py", line 257, in start
    main(parse_args(sys.argv[1:]))
  File "/Users/kuipersm/Library/Enthought/Canopy/edm/envs/User/lib/python3.5/site-packages/jaws/jaws.py", line 244, in main
    dispatch_converter(args, input_file, output_file, stations)
  File "/Users/kuipersm/Library/Enthought/Canopy/edm/envs/User/lib/python3.5/site-packages/jaws/jaws.py", line 213, in dispatch_converter
    converters[char](args, input_file, output_file, stations)
  File "/Users/kuipersm/Library/Enthought/Canopy/edm/envs/User/lib/python3.5/site-packages/jaws/imau2nc.py", line 188, in imau2nc
    common.load_dataset_attributes(sub_type, ds, args)
  File "/Users/kuipersm/Library/Enthought/Canopy/edm/envs/User/lib/python3.5/site-packages/jaws/common.py", line 151, in load_dataset_attributes
    ds[key1].attrs = value2.items()
  File "/Users/kuipersm/Library/Enthought/Canopy/edm/envs/User/lib/python3.5/site-packages/xarray/core/dataset.py", line 874, in __getitem__
    return self._construct_dataarray(key)
  File "/Users/kuipersm/Library/Enthought/Canopy/edm/envs/User/lib/python3.5/site-packages/xarray/core/dataset.py", line 783, in _construct_dataarray
    self._variables, name, self._level_coords, self.dims)
  File "/Users/kuipersm/Library/Enthought/Canopy/edm/envs/User/lib/python3.5/site-packages/xarray/core/dataset.py", line 77, in _get_virtual_variable
    ref_var = variables[ref_name]
KeyError: 'tilt_direction'
kuipersm@:~/Data/AWS/Antarctica/AWS17> jaws --rigb -4 -o ant_aws17_final_year2011_TEST.nc ant_aws17_goodangles.txt
Traceback (most recent call last):
  File "/Users/kuipersm/Library/Enthought/Canopy/edm/envs/User/lib/python3.5/site-packages/xarray/core/dataset.py", line 780, in _construct_dataarray
    variable = self._variables[name]
KeyError: 'tilt_direction'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/kuipersm/Library/Enthought/Canopy/edm/envs/User/bin/jaws", line 11, in <module>
    sys.exit(start())
  File "/Users/kuipersm/Library/Enthought/Canopy/edm/envs/User/lib/python3.5/site-packages/jaws/jaws.py", line 257, in start
    main(parse_args(sys.argv[1:]))
  File "/Users/kuipersm/Library/Enthought/Canopy/edm/envs/User/lib/python3.5/site-packages/jaws/jaws.py", line 244, in main
    dispatch_converter(args, input_file, output_file, stations)
  File "/Users/kuipersm/Library/Enthought/Canopy/edm/envs/User/lib/python3.5/site-packages/jaws/jaws.py", line 213, in dispatch_converter
    converters[char](args, input_file, output_file, stations)
  File "/Users/kuipersm/Library/Enthought/Canopy/edm/envs/User/lib/python3.5/site-packages/jaws/imau2nc.py", line 188, in imau2nc
    common.load_dataset_attributes(sub_type, ds, args)
  File "/Users/kuipersm/Library/Enthought/Canopy/edm/envs/User/lib/python3.5/site-packages/jaws/common.py", line 151, in load_dataset_attributes
    ds[key1].attrs = value2.items()
  File "/Users/kuipersm/Library/Enthought/Canopy/edm/envs/User/lib/python3.5/site-packages/xarray/core/dataset.py", line 874, in __getitem__
    return self._construct_dataarray(key)
  File "/Users/kuipersm/Library/Enthought/Canopy/edm/envs/User/lib/python3.5/site-packages/xarray/core/dataset.py", line 783, in _construct_dataarray
    self._variables, name, self._level_coords, self.dims)
  File "/Users/kuipersm/Library/Enthought/Canopy/edm/envs/User/lib/python3.5/site-packages/xarray/core/dataset.py", line 77, in _get_virtual_variable
    ref_var = variables[ref_name]
KeyError: 'tilt_direction'

pkuipersmunneke, 2018-12-20:
In the input file, I spotted a few tilt angle values of 180.0, 0.0 and -9999., but even after replacing these with a plausible number (e.g., 88.8) I get error this message.

ajcse1, 2018-12-20:
Thanks for sending the file @pkuipersmunneke . We were able to successfully convert this file on Mac OS Mojave, Python 3.5 and installing _jaws_ using pip.

Two things to note here:
1. I would recommend using anaconda instead of pip to install _jaws_, because anaconda handles dependencies of different packages well.
2. Please use Python 2.7, 3.6 or 3.7 only.

If you haven't, please install either [anaconda](https://conda.io/docs/user-guide/install/macos.html) or [miniconda](https://conda.io/miniconda.html) (Miniconda is a lighter version of anaconda).

Here's a [cheat-sheet](https://conda.io/docs/_downloads/conda-cheatsheet.pdf) for conda which might come handy in future.

Then you can create a new environment as following:
`conda create --name py37 python=3.7`

Then activate the newly created environment:
`source activate py37`

Then install jaws:
`conda install -c conda-forge jaws`

Please let me know if this resolves the error.

ajcse1, 2018-12-20:
> Interestingly, if I select only the uppermost 10 lines, I get
> 
> kuipersm@:~/Data/AWS/Antarctica/AWS17> jaws --rigb -4 -o ant_aws17_final_year2011_TEST.nc ant_aws17_goodangles.txt
> Traceback (most recent call last):
> File "/Users/kuipersm/Library/Enthought/Canopy/edm/envs/User/lib/python3.5/site-packages/xarray/core/dataset.py", line 780, in _construct_dataarray
> variable = self._variables[name]
> KeyError: 'tilt_direction'
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
> File "/Users/kuipersm/Library/Enthought/Canopy/edm/envs/User/bin/jaws", line 11, in 
> sys.exit(start())
> File "/Users/kuipersm/Library/Enthought/Canopy/edm/envs/User/lib/python3.5/site-packages/jaws/jaws.py", line 257, in start
> main(parse_args(sys.argv[1:]))
> File "/Users/kuipersm/Library/Enthought/Canopy/edm/envs/User/lib/python3.5/site-packages/jaws/jaws.py", line 244, in main
> dispatch_converter(args, input_file, output_file, stations)
> File "/Users/kuipersm/Library/Enthought/Canopy/edm/envs/User/lib/python3.5/site-packages/jaws/jaws.py", line 213, in dispatch_converter
> converters[char](args, input_file, output_file, stations)
> File "/Users/kuipersm/Library/Enthought/Canopy/edm/envs/User/lib/python3.5/site-packages/jaws/imau2nc.py", line 188, in imau2nc
> common.load_dataset_attributes(sub_type, ds, args)
> File "/Users/kuipersm/Library/Enthought/Canopy/edm/envs/User/lib/python3.5/site-packages/jaws/common.py", line 151, in load_dataset_attributes
> ds[key1].attrs = value2.items()
> File "/Users/kuipersm/Library/Enthought/Canopy/edm/envs/User/lib/python3.5/site-packages/xarray/core/dataset.py", line 874, in **getitem**
> return self._construct_dataarray(key)
> File "/Users/kuipersm/Library/Enthought/Canopy/edm/envs/User/lib/python3.5/site-packages/xarray/core/dataset.py", line 783, in _construct_dataarray
> self._variables, name, self._level_coords, self.dims)
> File "/Users/kuipersm/Library/Enthought/Canopy/edm/envs/User/lib/python3.5/site-packages/xarray/core/dataset.py", line 77, in _get_virtual_variable
> ref_var = variables[ref_name]
> KeyError: 'tilt_direction'
> kuipersm@:~/Data/AWS/Antarctica/AWS17> jaws --rigb -4 -o ant_aws17_final_year2011_TEST.nc ant_aws17_goodangles.txt
> Traceback (most recent call last):
> File "/Users/kuipersm/Library/Enthought/Canopy/edm/envs/User/lib/python3.5/site-packages/xarray/core/dataset.py", line 780, in _construct_dataarray
> variable = self._variables[name]
> KeyError: 'tilt_direction'
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
> File "/Users/kuipersm/Library/Enthought/Canopy/edm/envs/User/bin/jaws", line 11, in 
> sys.exit(start())
> File "/Users/kuipersm/Library/Enthought/Canopy/edm/envs/User/lib/python3.5/site-packages/jaws/jaws.py", line 257, in start
> main(parse_args(sys.argv[1:]))
> File "/Users/kuipersm/Library/Enthought/Canopy/edm/envs/User/lib/python3.5/site-packages/jaws/jaws.py", line 244, in main
> dispatch_converter(args, input_file, output_file, stations)
> File "/Users/kuipersm/Library/Enthought/Canopy/edm/envs/User/lib/python3.5/site-packages/jaws/jaws.py", line 213, in dispatch_converter
> converters[char](args, input_file, output_file, stations)
> File "/Users/kuipersm/Library/Enthought/Canopy/edm/envs/User/lib/python3.5/site-packages/jaws/imau2nc.py", line 188, in imau2nc
> common.load_dataset_attributes(sub_type, ds, args)
> File "/Users/kuipersm/Library/Enthought/Canopy/edm/envs/User/lib/python3.5/site-packages/jaws/common.py", line 151, in load_dataset_attributes
> ds[key1].attrs = value2.items()
> File "/Users/kuipersm/Library/Enthought/Canopy/edm/envs/User/lib/python3.5/site-packages/xarray/core/dataset.py", line 874, in **getitem**
> return self._construct_dataarray(key)
> File "/Users/kuipersm/Library/Enthought/Canopy/edm/envs/User/lib/python3.5/site-packages/xarray/core/dataset.py", line 783, in _construct_dataarray
> self._variables, name, self._level_coords, self.dims)
> File "/Users/kuipersm/Library/Enthought/Canopy/edm/envs/User/lib/python3.5/site-packages/xarray/core/dataset.py", line 77, in _get_virtual_variable
> ref_var = variables[ref_name]
> KeyError: 'tilt_direction'

It's happening because '--rigb' option is used and there is no clear day in those first 10 lines and hence there's no tilt_angle and tilt_direction. However, the program is still trying to attach the attributes for tilt_angle and tilt_direction. It's a bug and I am working to fix it.

ajcse1, 2018-12-27:
@pkuipersmunneke, please let us know whenever you have an update about the issue

closed 2019-03-07 by ajcse1