jaws/jaws #145: Sample data must work with RIGB if possible

czender, 2019-03-05:
Currently the GCNet sample data does not work with RIGB. Have not tested the others. The sample data must always work with RIGB for networks that have radiation. If a longer dataset (e.g., 1 month is necessary) then provide a sample dataset that is the minimal size necessary to get RIGB to work. Also, he error message below is inscrutable by me. Please make the most common errors have clear English descriptions with HINTs on how to avoid them.

```
(nvr_jaws) zender@firn:~$ jaws -D 0 ~/jaws/sample_data/GCNet_Summit_20140601.txt
(nvr_jaws) zender@firn:~$ jaws -D 0 --rigb ~/jaws/sample_data/GCNet_Summit_20140601.txt
20140531
Traceback (most recent call last):
  File "/Users/zender/anaconda/envs/nvr_jaws/bin/jaws", line 10, in <module>
    sys.exit(start())
  File "/Users/zender/anaconda/envs/nvr_jaws/lib/python3.7/site-packages/jaws/jaws.py", line 261, in start
    main(parse_args(sys.argv[1:]))
  File "/Users/zender/anaconda/envs/nvr_jaws/lib/python3.7/site-packages/jaws/jaws.py", line 248, in main
    dispatch_converter(args, input_file, output_file, stations)
  File "/Users/zender/anaconda/envs/nvr_jaws/lib/python3.7/site-packages/jaws/jaws.py", line 217, in dispatch_converter
    converters[char](args, input_file, output_file, stations)
  File "/Users/zender/anaconda/envs/nvr_jaws/lib/python3.7/site-packages/jaws/gcnet2nc.py", line 174, in gcnet2nc
    ds = tilt_angle.main(ds, latitude, longitude, clr_df, args)
  File "/Users/zender/anaconda/envs/nvr_jaws/lib/python3.7/site-packages/jaws/tilt_angle.py", line 102, in main
    fsds_intrp = CubicSpline(hours_nonmsng, fsds_jaws_nonmsng, extrapolate=True)(half_hours)
  File "/Users/zender/anaconda/envs/nvr_jaws/lib/python3.7/site-packages/scipy/interpolate/_cubic.py", line 531, in __init__
    raise ValueError("`x` must contain at least 2 elements.")
ValueError: `x` must contain at least 2 elements.
(

```

ajcse1, 2019-03-06:
In the RIGB section on Homepage, there is a link for separate sample GCNet file (gcnet_summit_20120817.txt) that works with --rigb option. I will include clear-sky day sample files only for all networks.

The error above has been fixed in the v0.7.3 released yesterday. Please update to the latest version.

czender, 2019-03-06:
OK but please remove GCNet_Summit_20140601.txt. If a network can support RIGB then all its sample files must be RIGB-compatible.

ajcse1, 2019-03-06:
Yeah

ajcse1, 2019-03-07:
Done

closed 2019-03-07 by ajcse1