jaws/jaws #166: Analysis plots error

ajcse1, 2019-04-03:
`
$ jaws --anl=monthly --var=temperature_cs500_1 --anl_yr=2012 --anl_mth=8
GCNet_Summit_20120817.nc
fails with “ERROR: Provide data for each day of the month”
`

ajcse1, 2019-04-03:
The sample file that we have provided contains data for only 1 day. To run 'monthly' and 'annual' analysis, user needs to have data for a month and year respectively.
Since these data are owned by different network managers, we don't have permission to distribute data for more than a day.

czender, 2019-04-03:
Recall that we now distribute a single month of data for networks that can use RIGB (right? so that users can try RIGB on the demo data). Please change the analysis demos to use the same monthly datasets as input.

ajcse1, 2019-04-03:
For RIGB, we need a clear-sky day. So, we are still distributing a single day data but that day is now clear-sky.

czender, 2019-04-03:
OK. Then just warn on the homepage which examples will only work with more data than the default.

ajcse1, 2019-04-03:
Yes, updated the instructions

closed 2019-04-08 by ajcse1