jaws/jaws #191: Add helpful HINTS to get JAWS running

czender, 2019-06-25:
Here is what happens when I, a non-Python person, try to run jaws:
```
zender@sastrugi:~/jaws/jaws$ python ./jaws.py
Traceback (most recent call last):
  File "./jaws.py", line 11, in <module>
    import gcnet2nc, promice2nc, aaws2nc, imau2nc, scar2nc, common, analysis
  File "/Users/zender/jaws/jaws/gcnet2nc.py", line 1, in <module>
    from metpy.units import units
ImportError: No module named metpy.units
```
I think JAWS has a perfect record of never working for me from the command line. Please make it fail with a helpful error message of _what I should type_ to get it running.

ajcse1, 2019-06-26:
As far as I have debugged for fresh installation on all platforms and Python version 2.7, 3.6, 3.7 with both conda and pip installations, this error only comes up with pip installation on Python 2.7.

The error is caused due to a dependent package 'pint'. They identified this [issue](https://github.com/conda-forge/metpy-feedstock/issues/27) and have corrected it but they only updated the conda-forge package after this update and not their pip package. I have contacted them regarding this.

On my end, when I encountered this error, after that I did `conda install jaws` and the error was gone.
Can you please try this? Also, can you please confirm your Python version using:
`python --version`

czender, 2019-06-26:
```
zender@sastrugi:~/jaws/jaws$ python --version
Python 2.7.10
```
The executable from the Conda package works for me. The executable from the Github repository is what gives this error. I am asking that you replace the inscrutable error with a HINT that tells the user _what to type_ to get jaws working. So the user of a Github-based jaws does not give up. Is that clear?

ajcse1, 2019-06-26:
Alright. Got it!

closed 2019-09-25 by ajcse1