jaws/jaws #199: RIGB Query

nithya255, 2020-06-19:
HI I have been trying to run RIGB in a sample dataset. The program ran successfully but I cannot find the output file. I know this is silly to ask but please guide in getting the output of the RIGB program

czender, 2020-06-19:
If you ran` jaws --rigb in.txt` then the output file would b enamed in.nc by default.

nithya255, 2020-06-19:
Good evening,
   But I could not find any .nc file. I had run the above program it was
successful still I am unable to find it

czender, 2020-06-19:
@wenshanw please help

wenshanw, 2020-06-19:
Hi @nithya255, would you please send me your sample dataset? 
Also, what operating system are you using? How/where did you install rigb?

nithya255, 2020-06-21:
Good Evening Everyone,
                I would like to borrow some time to run the codes again and
see if i am getting any .nc file. So that I can revert to you with the
results.

Thanks and Regards
Nithya Jeevanand

nithya255, 2020-06-22:
> Good Morning Jaws/Jaws,
>
>          I have used your sample data which I have attached below.  Today
> morning I tried the example codes you have mentioned on your website. I
> have used the RIGB command "$ jaws --rigb ABC.txt" and I modified it
> according to the sample data but still I am unable to find the output file
> in the destined path. Can you please revert as soon as possible?
>
> Thanks and Regards
> Nithya Jeevanand
>
> On Mon, Jun 22, 2020 at 10:45 AM Nithya Jeevanand <nithya25594@gmail.com>

Good Morning Jaws/Jaws,

         I have used your sample data from your website, the data was from
PROMICEdata portal. Today morning, I tried the example codes you have
mentioned on your website. I have used the RIGB command "$ jaws --rigb
ABC.txt" and I modified it according to the sample data but still I am
unable to find the output file in the destined path. Can you please revert
as soon as possible?
Thanks and Regards
NJ

wenshanw, 2020-06-22:
Hi @nithya255,

Would you please try "jaws -D 5 --rigb ABC.txt" and paste the output here?
Also, what operating system are you using? How/where did you install rigb?

nithya255, 2020-06-23:
Good Morning,
   Thank you for the reply, I have got the result and pasted it above. I am
using windows 7 professional operating system and I am using jaws in
anaconda but there is no instruction given how to download rigb in conda or
python nor any other platform in your website. If I have to download rigb
package please do guide me. Still I am unable to find the .nc file where it
is stored?
[image: image.png]

Thanks and Regards
NJ

nithya255, 2020-06-23:
Thank you for your response. I found the output file

wenshanw, 2020-06-23:
Hi @nithya255,

Where did you find the file? I wonder if we can revise the document accordingly to help future users.

As you found out, we don't need to install rigb separately. Once you install jaws, the `--rigb` option is included.

nithya255, 2020-06-23:
Good evening,

         It was found in my path of the input file. For example my input
file was in desktop "C:/Users/admin/Desktop/abc.txt" so I found the output
file in this "C:/Users/admin".

Thanks and regards
Nithya Jeevanand

nithya255, 2020-07-01:
GoodMorning,
        Hi, I have my own data for processing rigb but I want to know
regarding the input format and the necessary data needed in the input file.
Can you help me with it?

Thanks and regards
NJ

czender, 2020-08-11:
NJ,
This is documented at https://jaws.readthedocs.io/en/latest/
Charlie

closed 2021-12-29 by nithya255