jaws/jaws #201: Standards for short names

mankoff, 2020-09-30:
We're doing some re-coding of the PROMICE processing toolchain, and have the opportunity to make compatibility-breaking changes to the file formats, including for example using CF-compliant `standard_name`. My question is where JAWS gets the list of short-names.

For example, from https://github.com/jaws/jaws/blob/master/jaws/resources/promice/original_columns.json it appears that what PROMICE calls `ShortwaveRadiationDown(W/m2)`, JAWS calls `fsds`.

Elsewhere, IPCC has CMIP short name standards (see https://pcmdi.llnl.gov/mips/cmip3/variableList.html#Highest_priority_output) that define `surface_downwelling_shortwave_flux_in_air` as `rsds`.

Where does the JAWS `fsds` come from?

Related:  https://github.com/GEUS-PROMICE/PROMICE-AWS-processing/issues/2

czender, 2020-09-30:
The short variable names like `fsds` are employed (in ALL CAPS) in the CESM CAM and E3SM EAM climate models. We use those models heavily so it made sense to match those to the short names.  Perhaps we should add the CMIP names in a new attribute `cmip_name`. As you know, JAWS stores the original field name in `original_name`, and the CF standard name in `standard_name`. The important thing about the short name is that it be memorable and easy to type. For us, CAM/EAM names fit that criteria better than CMIP names.

mankoff, 2020-09-30:
Thanks for the explanation. We'll pick a standard short name (TBD) and I'll document it here https://github.com/GEUS-PROMICE/PROMICE-AWS-processing/issues/2

closed 2020-09-30 by mankoff