jaws/jaws #203: UnicodeDecodeError

s-jhen, 2021-01-07:
Plotting netCDF returning **UnicodeDecodeError**. JAWS command turning file into Windows-1254 encoding instead of default UTF-8. Not sure how to fix this, or why this is happening. 

Running Python 3.6.0

Cheers

czender, 2021-01-08:
Please post the exact command that generates this behavior so we can attempt to reproduce it.

s-jhen, 2021-01-11:
Let me know if you need something else, but this is what is happening.

**command**: jaws -a diurnal -v ta AAWS_AGO-4_20161130.nc

**returns**: 
Traceback (most recent call last):
  File "/Users/shendry/miniconda3/bin/jaws", line 8, in <module>
    sys.exit(start())
  File "/Users/shendry/miniconda3/lib/python3.8/site-packages/jaws/jaws.py", line 310, in start
    main(parse_args(sys.argv[1:]))
  File "/Users/shendry/miniconda3/lib/python3.8/site-packages/jaws/jaws.py", line 282, in main
    output_file = get_output_file(args, input_file, stations)
  File "/Users/shendry/miniconda3/lib/python3.8/site-packages/jaws/jaws.py", line 213, in get_output_file
    if stream.readline()[0] == 'D':
  File "/Users/shendry/miniconda3/lib/python3.8/codecs.py", line 322, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x89 in position 0: invalid start byte

Thanks

s-jhen, 2021-01-11:
There are no issues with text files.